arduino/Arduino #11552: Library Manager index is not updated

jandrassy, 2021-05-22:
I made a release yesterday and it is still not in Library Manager.
on https://www.arduinolibraries.info/ last addition is 20 hours ago

per1234, 2021-05-22:
Hi @jandrassy is it this one?:
https://github.com/jandrassy/EthernetENC/tags

The latest tag is 2.0.1 and that one is in the Library Manager index:
[image]

jandrassy, 2021-05-22:
Per, tag is 2.0.10

per1234, 2021-05-22:
OK, thanks for the update. The Library Manager indexer is currently undergoing maintenance so no new releases are being picked up. I'll post an update here once it is back in action.

jandrassy, 2021-05-23:
it works now

closed 2021-05-23 by jandrassy