arduino/Arduino #11555: Library Manager] Request to add to Library Manager: MS5x

abishur, 2021-05-24:
Please add my library to the library manager for the MS5xx, MS6xx, MS7xx, and MS8xx line of pressure sensors

https://github.com/abishur/arduino-ms5x

per1234, 2021-05-25:
Hi @abishur. Thanks for your interest in contributing to the Arduino Library Manager index.

Before we can add your library to Library Manager, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

abishur, 2021-05-26:
I h ave added the release that matches the tag in my library.properties file, thanks!

cmaglie, 2021-05-28:
Done.

closed 2021-05-28 by cmaglie