arduino/Arduino #11557: [Library Manager] Request to add to Library Manager

GyverLibs, 2021-05-25:
Hello! I'am known russian developer and Arduino popularizer, and I have a bunch of libraries. It would be great to add them to Library Manager.

Display:
https://github.com/GyverLibs/SevenSegmentsDisp
https://github.com/GyverLibs/GyverTM1637
https://github.com/GyverLibs/GyverOLED
https://github.com/GyverLibs/GyverMAX7219

Sensors:
https://github.com/GyverLibs/microDS18B20
https://github.com/GyverLibs/microDS3231
https://github.com/GyverLibs/Psychrometer
https://github.com/GyverLibs/Tachometer
https://github.com/GyverLibs/GyverNTC
https://github.com/GyverLibs/GyverHX711
https://github.com/GyverLibs/GyverBME280

Hardware input:
https://github.com/GyverLibs/EncButton
https://github.com/GyverLibs/NecDecoder
https://github.com/GyverLibs/AnalogKey
https://github.com/GyverLibs/GyverButton
https://github.com/GyverLibs/GyverEncoder

Hardware output:
https://github.com/GyverLibs/GRGB
https://github.com/GyverLibs/microLED
https://github.com/GyverLibs/ServoSmooth
https://github.com/GyverLibs/SoftServo
https://github.com/GyverLibs/GyverRelay
https://github.com/GyverLibs/PWMrelay
https://github.com/GyverLibs/GyverStepper
https://github.com/GyverLibs/GyverMotor
https://github.com/GyverLibs/GyverDimmer
https://github.com/GyverLibs/AccelMotor

AVR periphery:
https://github.com/GyverLibs/GyverWDT
https://github.com/GyverLibs/microWire
https://github.com/GyverLibs/GyverTimers
https://github.com/GyverLibs/directTimers
https://github.com/GyverLibs/directADC

Algorithms:
https://github.com/GyverLibs/GyverFilters
https://github.com/GyverLibs/GyverPID
https://github.com/GyverLibs/mString
https://github.com/GyverLibs/VolAnalyzer
https://github.com/GyverLibs/Random16
https://github.com/GyverLibs/Hamming
https://github.com/GyverLibs/TimeRandom
https://github.com/GyverLibs/UnixTime
https://github.com/GyverLibs/GyverGFX
https://github.com/GyverLibs/GyverFIFO
https://github.com/GyverLibs/Forecaster
https://github.com/GyverLibs/fixed
https://github.com/GyverLibs/FFT_C
https://github.com/GyverLibs/BitPack
https://github.com/GyverLibs/Clap
https://github.com/GyverLibs/CRT

Communication:
https://github.com/GyverLibs/FastBot
https://github.com/GyverLibs/Gyver433
https://github.com/GyverLibs/GyverBus
https://github.com/GyverLibs/GParser
https://github.com/GyverLibs/AsyncStream

System:
https://github.com/GyverLibs/GyverPWM
https://github.com/GyverLibs/GyverTimer
https://github.com/GyverLibs/GyverUART
https://github.com/GyverLibs/TimerMs
https://github.com/GyverLibs/GyverPower
https://github.com/GyverLibs/GyverOS
https://github.com/GyverLibs/buildTime

cmaglie, 2021-05-28:
Done, whew!

closed 2021-05-28 by cmaglie

GyverLibs, 2021-05-29:
Many thanks!