arduino/Arduino #11561: Request to add library to library manager: M5STACK_4RELAY

sfekilou, 2021-05-26:
Library to control M5Stack 4 Relay Unit

Here's the link to the library:
https://github.com/sfekilou/M5STACK_4RELAY
Thank you!

per1234, 2021-05-26:
Hi @sfekilou. Thanks for your interest in contributing to the Arduino Library Manager index.

Your library looks to be all correct to me. However, it was not correct at the time of the V1.0.0 tag due to missing the required library.properties file:
https://github.com/sfekilou/M5STACK_4RELAY/tree/V1.0.0

The Arduino Library Manager works exclusively from tags, so it's necessary for the library's latest tag to be valid in order for it to be accepted into the index.

So you'll either need to update the `version` value in library.properties and then create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) or else redo the existing tag.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

sfekilou, 2021-05-26:
Thanks,
Sorry, this was my first library publication on github.
Correct now ?

cmaglie, 2021-05-28:
Done.

closed 2021-05-28 by cmaglie