arduino/Arduino #11563: [Library Manager] Please add UncleRus-drivers

lbernstone, 2021-05-26:
This is a collection of drivers collected and maintained by [@UncleRus](https://github.com/UncleRus/esp-idf-lib) for ESP32.  I only write the script to flatten it for use in Arduino. It includes drivers for ads1115, bh1750, bme680, bmp180, bmp280, button, ccs811, dht, ds1302, ds1307, ds18x20_multi, ds18x20_single, ds3231, ds3502, encoder, hd44780_gpio, hd44780_i2c, hd44780_i2c_scroll, hmc5883l, hx711, ina219, ina260, ina3221, led_effects, led_strip, led_strip_spi, lm75, max31725, max7219_7seg, max7219_8x8, mcp23017, mcp23s17, mcp342x, mcp4725, mcp960x, mcp9808, mhz19b, ms5611_i2c, pca9685, pcf8563, pcf8575, pcf8591, qmc5883l, sgp40, sht3x, sht4x, si7021, simplest_barometer, tca95x5, tda74xx, tsl2561, tsl2591_interrupt, tsl2591_simple, tsl4531, tsys01, ultrasonic, wiegand_reader.

per1234, 2021-05-26:
Hi @lbernstone. Is this the repository for the library you want to add?:
https://github.com/lbernstone/UncleRus-drivers

lbernstone, 2021-05-26:
Doh! Yes

cmaglie, 2021-05-28:
Done.

closed 2021-05-28 by cmaglie