arduino/Arduino #11572: Library installed from library-manager has a different name

kulbhushanchand, 2021-05-29:
Thanks for accepting my library [MCP4251](https://github.com/arduino/Arduino/issues/11570). It is now available in the library manager.

When installing this library ([MCP4251](https://github.com/kulbhushanchand/MCP4251)) from the **library manager**, it is installed under a different directory name (`arduino_984644`) than the original name (`MCP4251`).

**Expected** - `C:\Users\Kulbhushan\Documents\Arduino\libraries\MCP4251`
**Actual** - `C:\Users\Kulbhushan\Documents\Arduino\libraries\arduino_984644`

**Please note everything else works fine, the issue is with the name of the installed directory only.**

- **Library browser**

[image]

- **Installed directory**

[image]

Few other libraries (`USBHost`, `ArduinoUnit`, `PulseSensor Playground`) also has the similar issue.

Kindly guide me if I'm doing something wrong?

Thanks

per1234, 2021-05-29:
Hi @kulbhushanchand. This `arduino_nnnnnn` folder is the temporary staging folder the library is installed to before the IDE renames it to the folder matching the library.properties `name` value. If for some reason the Arduino IDE is blocked from creating that final installation folder then the library installation remains in the temporary folder name. I usually get this durig experimentations when I do a Library Manager update while I have the existing library installation open, causing it to be locked from overwrite by the IDE. That's not likely to be a common set of circumstances for a normal user to produce, though more likely for a library developer such as yourself.

There was a report at https://github.com/arduino/Arduino/issues/11203 of this being caused by having Dropbox configured to monitor the `libraries` folder. Do you have Dropbox or some other cloud storage service enabled for your `C:\Users\Kulbhushan\Documents\Arduino\libraries` folder?

kulbhushanchand, 2021-05-29:
Thanks @per1234 for your prompt attention to the issue and your expert advice.

As you have mentioned, I do have [`Mega`](https://mega.nz/) cloud storage service enabled in background for `My Documents` folder. So to test your advice, I did fresh install of my library with `Mega` process killed, and **the issue resolved**. Just to be extra sure, I re-tested fresh library installation again with `Mega` process running with `upload/download` paused, and **the issue persists**.

**So, definitely, it was the cloud storage service in the background causing the issue.**
As of now, totally exiting the background service during library install is a good workaround.

per1234, 2021-05-29:
You're welcome and thanks for getting back to us with an update.

Since we already have a report at https://github.com/arduino/Arduino/issues/11203 of this problem with incomplete Library Manager installations to directories managed by cloud storage services, I'll go ahead and close this one so we can be sure to consolidate all relevant discussion in one place.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-05-29 by per1234