arduino/Arduino #11574: SD reference, forum link equals OOPS when following

sterretjeToo, 2021-05-29:
Page https://www.arduino.cc/en/Reference/SD

has a [link](http://arduino.cc/forum/index.php/board,23.0.html) to the forum: "Corrections, suggestions, and new documentation should be posted to the Forum."

That results in an OOPS page.

Please fix and please check all reference pages; doing so on the backend will be a lot easier than users checking every single page

Thanks

mahendra1290, 2021-09-03:
hi, @per1234 is this issue up for grab?

per1234, 2021-09-03:
Hi @mahendra1290. Thanks for your interest in contributing to the Arduino project. This documentation content is not hosted in a public repository so there is no possibility for you to fix the error.

There is some other Arduino documentation content hosted in a public repository and you are welcome to contribute to that content. There is a tutorial here:
https://create.arduino.cc/projecthub/Arduino_Genuino/contribute-to-the-arduino-reference-af7c37

mahendra1290, 2021-09-03:
> Hi @mahendra1290. Thanks for your interest in contributing to the Arduino project. This documentation content is not hosted in a public repository so there is no possibility for you to fix the [...]

Got it thanks buddy

per1234, 2022-07-17:
Thanks so much for reporting this error @sterretjeToo. I did a search of all the documentation content and did not find any remaining references to this URL.

The previous link now redirecting to https://forum.arduino.cc/. Not ideal, but at least better than a 404 page.

closed 2022-07-17 by per1234