arduino/Arduino #11588: :warning: New Library Manager submission/update system! :warning:

per1234, 2021-06-07:
I'm pleased to announce that we have created a new system for submitting libraries to the Arduino Library Manager, and for requesting changes to your prior submissions.

The new system is intended to be more streamlined and more transparent. A bot runs checks on the submission to make sure  it meets [the requirements](https://github.com/arduino/library-registry/blob/main/FAQ.md#what-are-the-requirements-for-a-library-to-be-added-to-library-manager) and will automatically accept it once passing. In the event some changes are needed, there is immediate feedback providing guidance to bring the the submission into compliance.

Instructions for the new system are here:

https://github.com/arduino/library-registry#readme

We'll still be monitoring the process and providing assistance when needed, just as before.

As of now, Library submissions via the old system of submitting an issue here will no longer be accepted. Issues to submit bug reports or feature requests related to the Arduino IDE's Library Manager are always welcome in this issue tracker.

Related discussion on the Arduino forum here:
https://forum.arduino.cc/t/new-library-manager-submission-update-system/871710

khoih-prog, 2021-06-08:
Thanks for the great work @per1234 . 

The new system is so easy and efficient to use.
Just try with the new [**MBED_RPI_PICO_TimerInterrupt** Library](https://github.com/khoih-prog/MBED_RPI_PICO_TimerInterrupt) without any issue.

khoih-prog, 2021-06-08:
I'm amazed that the library was  included in Library Manager within 1 hour from creating the PR. Compared to more than 2 weeks sometimes using the old system :1st_place_medal: