arduino/Arduino #11590: Memory Leak in arduino IDE

platypushunter, 2021-06-09:
I was teaching a summer camp and a few computers ended up having ~10-12 arduino windows open for about 5 or 6 hours because kids dont know how to close tabs, and even after closing the programs windows gave memory errors, specifically that programs couldnt open because there wasn't enough free memory.

fr0sty1, 2021-06-18:
I'm experiencing a similar situation with Arduino IDE version 1.8.15.

Running on Linux Mint 19 the IDE eventually consumes significant amounts of memory. As this goes on longer it eventually starts consuming singificant CPU resources (this is perhaps garbage collector related).

[image]

After that the program dies with OOM errors:
```
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
	at javax.swing.text.GapContent.allocateArray(GapContent.java:94)
	at javax.swing.text.GapVector.resize(GapVector.java:214)
	at javax.swing.text.GapVector.shiftEnd(GapVector.java:229)
	at javax.swing.text.GapContent.shiftEnd(GapContent.java:345)
```

PaulStoffregen, 2021-06-19:
GapContent is the default data storage scheme used by JTextArea in the serial monitor window.  Sadly, it's horribly inefficient if your board transmits a lot of data at very high speed or over a long time with the serial monitor window open.

Nearly 2 years ago I put a lot of work into optimizing the serial monitor.  Teensy 4.0 is able to transmit much faster than other boards, which was quickly crashing the IDE when anyone would upload a simple program which uses Serial.print() without delays.  I wrote a blog article about it at the time...

https://www.pjrc.com/improving-arduino-serial-monitor-performance/

I created a FifoDocument class which replaces the default Document & GapContent with highly optimized and fixed-size buffer scheme.  We've been using it since then at the default serial monitor implementation for Teensy boards.  All this code is open source and available on my github repositories, if anyone wants the code.

Unfortunately, integrating FifoDocument isn't a simple drop-in replacement.  It requires changes in the serial monitor code too, where incoming data is normally copied.  Even with a fixed-size FIFO, the Java garbage collection just can't keep up if the serial monitor abstraction layers make copies of data while it's arriving at sustained 20 Mbyte/sec speed.  This might be less of an issue with lower speeds, but my main design goal was to handle high speed data.

If Arduino is ever interested to integrate FifoDocument, I'd be happy to work with them.  We have exchanged some messages about it, so they are aware of my serial monitor optimization work.  My overall impression is they're primarily focused on developing the new 2.0 IDE in Go language.  Unless there is strong interest expressed by Arduino, I'm not going to do much more on contributing FifoDocument, other than occasionally writing messages like this to offer some explanation of the problem and of course make the source code available.  It's all open source and available on my public repositories for anyone who wants it.  The code has lots of comments explaining how it works, and that blog article has a high-level summary of the design.