arduino/Arduino #11594: IDE usability Improvement

lmcreek, 2021-06-12:
My sugestion is as follows:
On menu item Programmers add an entry 'Default Arduino Programmer'
make IDE button -> (right arrow) use the Programmer selected to upload the program.

It would allow those like me to press the button while using USBASP, or anyother, to upload program.
Not sure if this is the right place, but...

per1234, 2021-06-12:
Hi @lmcreek. Thanks for taking the time to request this feature. You'll be glad to know that the Arduino boards platform system already provides a mechanism for causing the "Upload" button to act as "Upload Using Programmer":
https://arduino.github.io/arduino-cli/latest/platform-specification/#upload-using-programmer-by-default

If the boards platform you are using does not provide this configuration, you can add it via a custom boards platform. You can see an example of a nice way to do this in [MiniCore](https://github.com/MCUdude/MiniCore). A [**Tools > Bootloader**](https://github.com/MCUdude/MiniCore#bootloader-option) menu appears when you select a MiniCore board, provided by a [custom board option](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options). If you select "No Bootloader" option from the menu, the IDE's "Upload" button does an "Upload Using Programmer":
https://github.com/MCUdude/MiniCore/blob/v2.1.2/avr/boards.txt#L74-L78

closed 2021-06-12 by per1234

lmcreek, 2021-06-12:
right on, thanks