arduino/Arduino #11595: Library Manager Dependencies list goes off screen (bug)

float13, 2021-06-13:
For a library with a very long list of dependencies (e.g. Adafruit MLX90640, which has 30+ dependencies), the list goes off the bottom of the screen (on Mac OS X), and the actual "Install" buttons are not visible or accessible. 

I'm not able to resize the window to fix it either (clicking/dragging the edges doesn't work).

Adding a scrollbar & setting a default/max size on the window would probably help.

I haven't done much GUI programming in Java but happy to help if I can.

float13, 2021-06-13:
(Temporary workaround: press 'Tab', then use 3-finger down pull to show all windows; see which button is highlighted; repeat until 'Install All' is selected, then hit 'Enter' to install (for some reason 'Return' did not work, but Fn+Return = 'Enter' worked on Macbook Pro.)

per1234, 2021-06-13:
Hi @float13. Thanks for taking the time to submit an issue.

I see we already have a prior issue report about this at https://github.com/arduino/Arduino/issues/10647.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-06-13 by per1234