arduino/Arduino #11598: http://

blikkk, 2021-06-15:
Error compile:

`const char format_html[] PROGMEM = R"rawliteral(
<meta http-equiv="refresh" content="51; url=http://8.8.8.8/index.html">
)rawliteral";`

per1234, 2021-06-15:
Hi @blikkk. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there:

https://forum.arduino.cc/

closed 2021-06-15 by per1234

blikkk, 2021-06-15:
This is bug.

per1234, 2021-06-15:
This issue tracker is for reporting bugs in the Arduino IDE code base. There is nothing in the very limited information you have provided that indicates your problem is caused by the Arduino IDE code base. For this reason, it is best for us to continue the investigation over on the Arduino forum, where we are free to guide you through the process of gathering and reporting sufficient information to allow an investigation to even begin. After that, if it turns out that your problem really is caused by a bug in the Arduino IDE code base, I'll be happy to reopen this issue.

For now, I'll go ahead and lock the conversation since it is not appropriate to continue it here. I'll see you over on the forum!