arduino/Arduino #11623: [Documentation] Typo on library page

KillzoneKid, 2021-07-04:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
https://www.arduino.cc/en/Reference/ArduinoSound

Current:

```
To use this library
#include <AudioSound.h>
```

Expected

```
To use this library
#include <ArduinoSound.h>
```

Xanth0k1d, 2021-07-18:
Hi, might be irrelevant, but I saw you looking for firmwares of bt201, I could help you.
You can reach me at my email xanth0k1d@gmail.com, will explain the origins of bt201 etc.

KillzoneKid, 2021-11-18:
How long does it take to correct a simple typo? How do you expect people willing to contribute if you do nothing about it?

kengdahl, 2021-12-07:
@per1234 Was this issue fixed in arduino-libraries/ArduinoSound#39?

per1234, 2021-12-07:
No. This issue is about the error on https://www.arduino.cc/en/Reference/ArduinoSound and the error is still on that page. arduino-libraries/ArduinoSound#39 only fixed the error on the https://www.arduino.cc/reference/en/libraries/arduinosound/ reference that is automatically generated from the contents of the https://github.com/arduino-libraries/ArduinoSound repository.

I don't know much about how the library reference pages work, but I suspect that https://www.arduino.cc/en/Reference/ArduinoSound can only be fixed by editing it in pmWiki (or whatever the old documentation system is called).

kengdahl, 2021-12-09:
This should be fixed now. Sorry about the delay.

closed 2021-12-09 by kengdahl