arduino/Arduino #11625: RP2040 boards have inconsistent delay and micros() times.

greiman, 2021-07-11:
delay(ms) only delays for about 96% of the requested time on Pico and Nano RP2040.  micros() seems to run at the correct rate.

The following program demonstrates the problem.  loop() has a delay(1000) but executes in about 960 microseconds.
```
uint32_t start;
void setup() {
  Serial.begin(9600);
  while(!Serial) {}
  start = micros(); 
}
uint32_t count = 0;
void loop() {
  Serial.print(count++);  
  Serial.print(' ');  
  Serial.println((micros() - start)/1000000.0);
  delay(1000);
}
```
Here is output.  The first column is loop count and the second column is time in seconds.  The second column should be >= to the first column.

> 0 0.00
> 1 0.96
> 2 1.92
> 3 2.88
> 4 3.84
> 5 4.80
> 6 5.76
> 7 6.72
> 8 7.68
> 9 8.64
> 10 9.60
> ...
> 93 89.28
> 94 90.24
> 95 91.20
> 96 92.16
> 97 93.12
> 98 94.08
> 99 95.04
> 100 96.00

I suspect the problem is due the RP2040 having a CPU rate of 125 MHz but in:

D:\mbed-os-rp2040_pr\targets\TARGET_RASPBERRYPI\TARGET_RP2040\TARGET_NANO_RP2040_CONNECT\board.c

`uint32_t SystemCoreClock = 120000000;`

I think micros() returns the RP2040 hardware microsecond clock but delay() uses an mbed wait based on an incorrect SystemCoreClock.

dwetchells, 2021-07-11:
Hi Bill,  I would gamble to say this test is a bit unfair.  The instruction to print out to serial port does take some time to run from the process and your performing a math function.  My 2 cents.

greiman, 2021-07-11:
> Hi Bill,  I would gamble to say this test is a bit unfair.  The instruction to print out to serial port does take some time to run from the process and your performing a math function.  My 2 cents.

That's exactly the point I made.  The loop has a delay of 1000 ms but take less than 1000 ms as measured by micros().  This is impossible if delay() and micros() are correct since the loop plus print should take longer than 1 second but it take 0.96 seconds.

I first noticed this in a loop that had no print that completed in less time than the delays in the loop.

fjansson, 2022-07-17:
I think this is the same as https://github.com/arduino/ArduinoCore-mbed/issues/271 and that it has been fixed by defining the right clock frequency in mbed os: https://github.com/arduino/mbed-os/commit/3d5c553f2

closed 2022-07-17 by per1234