arduino/Arduino #11638: Problem Arduino Nano Connect UF2 blink file not working

bonnom, 2021-07-26:
I don't know if this is the correct place to place a bug report but I couldn't find another place to do so.
Some people have problems with uploading the blink UF2 file from the 'Nano RP2040 Connect technical reference' https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-01-technical-reference.

I think this is caused by an undocumented change of the flash chip. According to the documents the Arduino Nano Connect should have a AT25SF128A-MHB-T. However newer boards have an ISSI IS25LP128F-JKLE. A completely different flash chip.

Circuitpython also didn’t use to work but than they added IS25LP128F-JKLE and now theirs UF2 file works!

Could the blink UF2 file and the documentation be updated so that it does take the IS25LP128F-JKLE into account?

pnndra, 2021-08-16:
Hi,
Sorry for the slow reply. Due to the (not so) recent issues with supply chain we're mounting different flash chips depending on availability. While those are mostly compatible with each other, some have slight changes that require fw adjustments. In this specific case the issue can be solved by compiling sketches using the latest core which is of course backwards compatible with all other chips. If you already flashed a board with code compiled with an old core you need to recover by resetting the board while shorting the rec pin with ground. Please make sure you remove the short after reset, before flashing.

bonnom, 2021-08-30:
I managed to fix the issue, but other might still get the problem!

mcejp, 2021-09-02:
Does 'the latest core' refer to _Mbed OS Nano Boards_ 2.4.1? I'm having the same issue, even a blinky sktech fails to boot after uploading.

iKK001, 2021-09-08:
You write:  `"...In this specific case the issue can be solved by compiling sketches using the latest core..."`:

How can I achieve this when working with a 3rd party library ?
Does the library creator need to implement the latest core first ?
Or can I do that by myself. If yes, how exactly (i.e. step by step explanation please).

I already tried:

1. I connect pin GND to pin RESET (or REC as you call it)
2. I power the board by plugging in the USB connector
--> I observe the RPI-RP2-drive cropping up on my desktop screen
3. I disconnect pin RESET from GND
4. I download the WiFiManager_NINA_Lite library's Example
i.e. in Arduino IDE under Examples --> WiFiManager_NINIA_Lite --> RP2040_WiFiNINA
5. At startup, still the same error :

```
Starting RP2040_WiFiNINA on MBED NANO_RP2040_CONNECT
WiFiManager_NINA_Lite v1.4.0
[WN] Hostname=RP2040-Master-Controller
LittleFS size (KB) = 64
LittleFS Mount Fail
Formatting...
```

Can you please explain better, step-by-step how you solve this problem with the library of concern ? Thank you.

pennam, 2021-09-08:
Hi @iKK001 regarding issue https://github.com/arduino/ArduinoCore-mbed/issues/318 you have already installed the latest core, so the new flash chip is already supported. I'm thinking you are facing a different issue otherwise your sketch wont even boot. What is the return value of the mbed mount function?

bonnom, 2021-09-08:
> Does 'the latest core' refer to _Mbed OS Nano Boards_ 2.4.1? I'm having the same issue, even a blinky sktech fails to boot after uploading.

Could very well be. Circuitpython had the exact same problem. When the new flash chip was added the configuration, the problem was solved.

iKK001, 2021-09-08:
Hi @pennam : How do I get the return value of the bed mount function ? (step by step explanation needed since I am not an expert)

pennam, 2021-09-08:
i will reply you here https://github.com/arduino/ArduinoCore-mbed/issues/318

iKK001, 2021-09-09:
[As @pennam  found out](https://github.com/arduino/ArduinoCore-mbed/issues/318#issuecomment-916350152), the problem can be solved by going back to the mbed-core version 2.3.1 (instead of 2.4.1).