arduino/Arduino #11641: Follow XDG Base Directory specification on Linux

erlisdhima, 2021-07-31:
By default the Arduino IDE creates a ~/.arduino15.

First issue, its on the $HOME dir. Ideally the [XDG Base Directory specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html) must be used.
Second, the directory naming is specific to some version? Why not just .arduino ?

Finally, the files/directories inside it are all in one place.

```
$ ls ~/.arduino15
cache
library_index.json
library_index.json.sig
logs
package_index.json
package_index.json.sig
preferences.txt
```

As per the specification, this should be something like below...

Config files go into $XDG_CONFIG_HOME
From all the files, only `preferences.txt` has configs.

Cache files go into $XDG_CACHE_HOME
In this case it should be the `cache` directory.

[actions history (logs, history, recently used files, …)](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html#variables) go into $XDG_STATE_HOME
Here I'm assuming everything else from that directory are state files, and the logs directory, well, for logs.

erlisdhima, 2021-07-31:
I found its doing this for all platforms. https://github.com/arduino/Arduino/blob/57a931c9c4c084057417837239ad8136f8a7b1aa/arduino-core/src/processing/app/Platform.java#L77-L81

Since java doesn't have a way to get the XDG dirs directly, I found this example https://github.com/omajid/xdg-java/blob/master/src/main/java/org/freedesktop/BaseDirectory.java

That does not include $XDG_STATE_HOME though, and it will need to be added.

per1234, 2021-07-31:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3915

closed 2021-07-31 by per1234