arduino/Arduino #11642: fatal java error when opening file open dialog (with mouse and ctrl o)

antonilol, 2021-08-02:
when i try to open the file open dialog the ide freezes and crashes after 2-3 seconds after freezing.
the following is then printed in the terminal:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fc04c7710cc, pid=1626, tid=1735
#
# JRE version: Java(TM) SE Runtime Environment (16.0.1+9) (build 16.0.1+9-24)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (16.0.1+9-24, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  0x00007fc04c7710cc
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/lib/systemd/systemd-coredump %P %u %g %s %t %c %h" (or dumping to /home/antoni/core.1626)
#
# An error report file with more information is saved as:
# /home/antoni/hs_err_pid1626.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
/usr/share/arduino/arduino: line 35:  1626 Aborted                 (core dumped) "$JAVA" "${JAVA_OPTIONS[@]}" processing.app.Base "$@"

```

antonilol, 2021-08-02:
btw i use arch

antonilol, 2021-08-02:
when i start it with java 8 it looks nicer and doest give me this crash
maybe something with java

facchinm, 2021-08-02:
Fellow archer :slightly_smiling_face: 
The problem is indeed JRE 16; the IDE has been validated only with JRE8 (the one we bundle) and it's known to be buggy on JRE9+ (may work to some extent and then crash during random operations).
My advice is to use the `archlinux-java` helper to set jre8 as default, so everything should work smoothly.

closed 2021-08-02 by facchinm