arduino/Arduino #11645: LiquidCrystal Library is missing Example "CustomCharacter"

TheOldBrick, 2021-08-09:
The LiquidCrystal library in the Arduino IDE provides several examples, including "CustomCharacter".  However, the "CustomCharacter" example is not described in the reference (https://www.arduino.cc/en/Reference/LiquidCrystal).

New users might be confused without proper documentation because the example requires two 10K potentiometers.

buddywhitman, 2021-12-26:
Can I work on this too, please?

per1234, 2022-04-20:
Thanks @karlsoderby! Should a link to the examples also be added to [the LiquidCrystal library reference home page](https://github.com/arduino-libraries/LiquidCrystal/blob/master/docs/readme.md), consistent with how it is done in some of the other library references ([example](https://github.com/arduino-libraries/Servo/tree/master/docs))?

As it is now, I think it will be difficult for the users researching the LiquidCrystal library to find the associated tutorials. I had a bit of trouble finding it in docs.arduino.cc myself even though I knew the approximate location from the path in the repository.

9500753429, 2024-10-18:
ill take this