arduino/Arduino #11647: Setting an Input Pin to High would be helpfull of a Warning of the Arduino IDE Software

DanielHaltiner, 2021-08-13:
Arduino IDE Version: 1.8.15
Arduino Hardware: Arduino Nano Every

If you write "digitalWrite (X, HIGH) to a Pin, which has accidentaly defined as an input the processor will set the output, but only a low current commes from this "Fake Output".

It would be very helpfull, if the IDE Software would give a Warning/Input, that in the code is a "digitalWrite" which writes to an Input...
(By defining the analog Inputs I defined the Digital 6 Output further below in the VoidSetup acidentaly again as an Input. This creates an Output half alive... Or I was to dumb to make right code)

[WCB-TRM21Software_V0.1.12.zip](https://github.com/arduino/Arduino/files/6981431/WCB-TRM21Software_V0.1.12.zip)

dsyleixa, 2021-09-02:
I double that, and as a  workaround I meanwhile use sth like
#define digitalWriteX(p,v) {pinMode(p, OUTPUT); digitalWrite(p,v);}

(similar things for digitalReadX(), digitalReadXpu(), digitalReadXpd()  )

hossians, 2021-09-03:
Writing a high to an input activates the Pull Up, and as I use that very frequently I would be annoyed by such a warning.

DanielHaltiner, 2021-09-03:
I understand you would be annoyed.
On the other hand to have a processor output just running fine without load (only Pull Up res), and with a bit load fails, is for probably most Arduino user a big big mystery… (yes, I made the initialisation wrong. But a hint what is wrong would help I guess many people).

Anyway, thanks for your excellent work about Arduino!

See you,
Daniel