arduino/Arduino #11664: Unwanted pulse warning for devices connected to onboard LED and corresponding pin

brycecherry75, 2021-09-07:
I have noted when the microcontroller RESET signal is negated, the Arduino bootloader flashes the onboard LED on a particular pin (which depending on the microcrontroller can support SCK as an SPI hardware function on this pin) and this occurrence (with an ATmega328P) was causing an HMC830 PLL/VCO (attached to hardware SPI pins with the SPI hardware function used) to latch an operating mode (according to the manufacturer's Operating Guide) when not wanted which can only be reset by removing power to the HMC830.

So therefore, I think there should be a warning about unwanted pulses (and possible unwanted interactions on connected hardware) on the pin connected to the onboard LED which flashes under bootloader control after negation of the RESET signal.

Swastyy, 2021-09-20:
Hi @brycecherry75 , @per1234 I want to work on this. Please assist and assign it to me.

per1234, 2021-09-20:
Hi @Swastyy. Thanks for your interest in contributing to the Arduino documentation. The relevant documentation is not hosted in a  public repository so you won't be able to contribute to it directly. You are welcome to comment here with a suggestion for the documentation content if you like.