arduino/Arduino #11671: How can I contribute to translate the Arduino Sensor Kit webpage into German language

Bodobolero, 2021-09-23:
I am professional software developer, native German speaker and Arduino fan.

I would like to help create the German translation for the sensor kit lessons

https://sensorkit.arduino.cc

Currently it has English, Spanish, and a few other languages but not German.
Is German already work in progress or can I help?
I did not find a GitHub repo for this documentation.

Bodobolero, 2021-10-03:
@kengdahl interested?

g-castronovo, 2021-12-07:
@Bodobolero Guten Morgen! I am Gaia and I am the localisation specialist at Arduino. Currently, Sensorkit content isn't open to the community for translations. The localisation in German isn't planned for 2021. However, I'll forward the request and see if there is interest for it. Nontheless, I'd like to thank you for your support!

kengdahl, 2021-12-21:
I will close this Issue since @Bodobolero has his answer.

closed 2021-12-21 by kengdahl