arduino/Arduino #11672: Arduino IDE deletes entire contents of build.path directory

bluepylons, 2021-09-30:
I set my build.path directory to my  C;\Users\<user>\Downloads folder, so that I could conveniently retrieve the .hex files if needed (There was a week-long period where the only AVR programmer I had was a cheap Chinese thing that used a proprietary program, and I needed the .hex files in a convenient spot to send via that). Unfortunately, I did not realize that Arduino IDE occasionally deletes the entire contents of its build path, and I lost the entirety of my Download folder's contents (which contained many, many, many non-Arduino files. including photographs, memes, old projects, many of which were not backed up). 

Can there either be:
1.) A smarter delete routine for the files in the build path so that it doesn't clear out any unrelated files, or:
2.) A huge warning to set your build path in a dedicated directory not shared with any other files.

per1234, 2021-09-30:
I think the appropriate place to document this behavior is alongside the documentation of the `build.path` preference:
https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#preferences

I believe this is the only place the preference is officially documented.

For this particular use case, the IDE's **Sketch > Export Compiled Binary** is a more user friendly approach.

bluepylons, 2021-09-30:
Thank you - I will do that in the future. 

Unfortunately I was following a [Sparkfun guide](https://learn.sparkfun.com/tutorials/wireless-arduino-programming-with-electric-imp/getting-hex) that I found off of a Google search, and any updates to the documentation warning of the potentially catastrophic consequences (e.g. someone setting it to their /My Documents/ folder that hasn't been backed up recently and losing all of their documents) probably won't propagate along to 3rd party articles. If changing the behavior isn't viable (as it would be a fair amount of work to implement a smarter delete routine), would it instead be possible to include a warning as a comment in preferences.txt itself? (though it appears that preferences.txt may not support comments)