arduino/Arduino #11673: Cannot Open Arduino

RockZombie123, 2021-09-30:
So recently, I have downloaded Arduino for Linux, Raspberry pi. I have successfully downloaded it, but can't open it. 
I can't take a screenshot, so I am going to try and describe it the best I can: Whenever I actually click on Arduino, it just shows a loading screen and then stops loading. It's probably crashing, but I need Arduino to somehow work.

per1234, 2021-10-01:
Hi @RockZombie123. In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to run the Arduino IDE from the command line, then share the output.

Please do this:

1. Start a terminal in the Arduino IDE installation folder
1. Type `./arduino`
1. Press **<kbd>Enter</kbd>**.
1. Wait for the Arduino IDE to finish starting, or failing to start.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Click and drag the mouse to select all the output in the terminal window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**. This will copy the selected text to the clipboard.
1. Open a forum reply here by clicking the **<kbd>Reply</kbd>** button.
1. Click the **`</>`** icon on the post composer toolbar. This will add the forum's code block markup (**` ``` `**) to your reply to make sure the output is correctly formatted.
   [image] 
1. Press **<kbd>Ctrl</kbd>+<kbd>V</kbd>**. This will paste the compilation output into the code block.
1. Move the cursor outside of the code block markup before you add any additional text to your reply.
1. Click the **<kbd>Reply</kbd>** button to post the output.

RockZombie123, 2021-10-01:
@per1234 When I tried `./arduino`, It gave me this error:

`./arduino: line 35: /home/pi/saish/programs/arduino-1.8.12/java/bin/java: cannot execute binary file: Exec format error`

RockZombie123, 2021-10-01:
https://github.com/arduino/Arduino/labels/Waiting%20for%20feedback

per1234, 2021-10-01:
OK, great. This error means the version of Arduino IDE you are using is not compatible with the architecture of your Raspberry Pi computer. It's necessary to use the appropriate download link from the Arduino Software page:
https://www.arduino.cc/en/software
The ARM Linux version of the Arduino IDE is the one for the Raspberry Pi. The "Linux ARM 32 bits" download link will usually be the correct one, but if you have installed a 64 bit operating system then you should use the "Linux ARM 64 bits" download link.

closed 2021-10-01 by per1234

RockZombie123, 2021-10-01:
`/home/pi/Downloads/arduino-nightly/arduino: line 35: /home/pi/Downloads/arduino-nightly/java/bin/java: cannot execute binary file: Exec format error`

per1234, 2021-10-01:
@RockZombie123 your comment doesn't provide any information other than that you are now using the hourly (AKA "nightly") build of the Arduino IDE.

If you have found a bug in the IDE, then you need to provide enough information for us to be able to investigate. If you want help, then you should post on the Arduino Forum instead:
https://forum.arduino.cc/

PaulStoffregen, 2021-10-01:
In a terminal, run "file /home/pi/Downloads/arduino-nightly/arduino-builder".  It should print this:

    arduino-builder: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped