arduino/Arduino #11688: corrupted files in \Arduino\portable\packages\arduino\tools ?

dsyleixa, 2021-10-20:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
trying to backup my Arduino folder I get the copy errors:
FEHLER (A) DATEINAME  D:\Arduino\portable\packages\arduino\tools\openocd\0.10.0-arduino7\share\openocd\scripts\target\?1879x?1?.cfg
FEHLER (A) DATEINAME  D:\Arduino\portable\packages\arduino\tools\openocd\0.10.0-arduino7\share\openocd\scripts\target\1986??1?.cfg

the files in question are actually:
1986ве1т.cfg
```
# 1986Ð’Ð•1Ð¢
# http://milandr.ru/index.php?mact=Products,cntnt01,details,0&cntnt01productid=236&cntnt01returnid=68

source [find target/swj-dp.tcl]

if { [info exists CHIPNAME] } {
   set _CHIPNAME $CHIPNAME
} else {
   set _CHIPNAME 1986Ð²Ðµ1Ñ‚
}

if { [info exists ENDIAN] } {
   set _ENDIAN $ENDIAN
} else {
   set _ENDIAN little
}

# Work-area is a space in RAM used for flash programming
if { [info exists WORKAREASIZE] } {
   set _WORKAREASIZE $WORKAREASIZE
} else {
   set _WORKAREASIZE 0x4000
}

#jtag scan chain
if { [info exists CPUTAPID] } {
   set _CPUTAPID $CPUTAPID
} else {
   if { [using_jtag] } {
      set _CPUTAPID 0x4ba00477
   } {
      # SWD IDCODE
      set _CPUTAPID 0x2ba01477
   }
}
swj_newdap $_CHIPNAME cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_CPUTAPID
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_m -endian $_ENDIAN -dap $_CHIPNAME.dap

# use AHB-Lite SRAM for work area
$_TARGETNAME configure -work-area-phys 0x20100000 -work-area-size $_WORKAREASIZE -work-area-backup 0

# can't handle overlapping memory regions
if { [info exists IMEMORY] && [string equal $IMEMORY true] } {
   flash bank ${_CHIPNAME}_info.flash mdr 0x00000000 0x01000 0 0 $_TARGETNAME 1 1 4
} else {
   flash bank $_CHIPNAME.flash mdr 0x00000000 0x20000 0 0 $_TARGETNAME 0 32 4
}

# JTAG speed should be <= F_CPU/6. F_CPU after reset is 8MHz, so use F_JTAG = 1MHz
adapter_khz 1000

adapter_nsrst_delay 100
if {[using_jtag]} {
   jtag_ntrst_delay 100
}

if {![using_hla]} {
   # if srst is not fitted use SYSRESETREQ to
   # perform a soft reset
   cortex_m reset_config sysresetreq
}

```

к1879xб1я.cfg
```
# Ð¡Ð‘Ð˜Ð¡ Ðš1879Ð¥Ð‘1Ð¯
# http://www.module.ru/catalog/micro/mikroshema_dekodera_cifrovogo_televizionnogo_signala_sbis_k1879hb1ya/

adapter_khz 1000

if { [info exists CHIPNAME] } {
    set _CHIPNAME $CHIPNAME
} else {
    set _CHIPNAME Ðº1879Ñ…Ð±1Ñ
}

if { [info exists ENDIAN] } {
    set _ENDIAN $ENDIAN
} else {
    set _ENDIAN little
}

if { [info exists DSP_TAPID] } {
    set _DSP_TAPID $DSP_TAPID
} else {
    set _DSP_TAPID 0x2b900f0f
}

jtag newtap $_CHIPNAME dsp -irlen 4 -expected-id $_DSP_TAPID

if { [info exists CPU_TAPID] } {
    set _CPU_TAPID $CPU_TAPID
} else {
    set _CPU_TAPID 0x07b76f0f
}

jtag newtap $_CHIPNAME arm -irlen 5 -expected-id $_CPU_TAPID

set _TARGETNAME $_CHIPNAME.arm
target create $_TARGETNAME arm11 -chain-position $_CHIPNAME.arm

```

are they using kyrillic chars in the file names? are they corrupted? can they be erased or are they important?

dsyleixa, 2021-10-21:
sorry, I don't understand:
what should I do exactly?  \Arduino\portable\package_index.json is what for?

cmaglie, 2021-10-21:
The file `1986ве1т.cfg` is part of OpenOCD, so it's normal for that file to be there.

`openocd/0.10.0-arduino12/share/openocd/scripts/target/1986ве1т.cfg`

BTW it's weird that your OS is not able to copy it, maybe the media you're using to backup is formatted with a filesystem that do not support UTF-8 characters?
Anyway you can safely ignore those files (unless you have a CPU based on 1986ве1т chipset but I don't think so...)