arduino/Arduino #11705: Serial Monitor fails to set 250000 baud rate with error message.

HoverClub, 2021-11-19:
Bug Report

Arduino 1.8.16
Linux Mint 19.2 (Ubuntu)

Arduino IDE serial monitor throws error "_Error while setting serial port parameters: 250,000 N 8 1" when setting baud rate to 250000_". It doesn't do like this when selecting any other baud rate up to 2000000.  With rates other than 250000 the monitor comms work OK.

If another App is run first using the 250000 baud rate on the same port then closed and the Arduino IDE is started, the serial comms all works as expected at 250000 (error message still displays though).

EricTra, 2023-04-29:
Hello there, here some details I got you

per1234, 2023-04-29:
@EricTra you are welcome to comment in the threads of Arduino's repositories when you have something of value to add to the discussion. However, the comments you made today didn't add anything. That type of comment only generates noise for the hundreds of people who are monitoring the repository to follow the project development and thus receive a notification for every comment. In aggregate, such noise can have a very harmful effect on free open source projects because the contributors will either waste time sorting through the unnecessary notifications, or else be forced to stop monitoring the conversations entirely.

It appears you are copy/pasting AI-generated content. This sort of content can be useful and appropriate in some cases, but only if used responsibly.

Just as with any other information you find on the Internet, you must carefully evaluate AI-generated content for accuracy, relevance, and appropriateness within the context. Copy/pasting without giving any thought to the information is irresponsible, regardless of the source of that information. If you are not able to make such an evaluation, then please don't use it in Arduino's repositories.

Thanks in advance for your cooperation.

EricTra, 2023-04-29:
> Thanks for give me feedback, I will learn from it next time
Have a good day 
Man