arduino/Arduino #11712: Linux, system wide installation isn't done correctly

avlap, 2021-12-01:
When running the install script as root, the *.desktop file is placed in /usr/share/applications. I think that should be /usr/local/share/applications or /opt/.....

It seems to be fixed in the new IDE install:
https://github.com/arduino/arduino-ide/issues/641
https://github.com/arduino/arduino-ide/pull/356/commits/b4db350ffbc80bf42a177a959f120e73c959a2f2

Processing seems to have this issue as well. Both install scripts are very similar. 
https://github.com/processing/processing4/issues/308#issuecomment-983098130

DragonflyRobotics, 2022-03-13:
So is this resolved?

avlap, 2022-03-13:
When I install both Arduino and Processing in /opt using the default install scripts, when Arduino launches, it displays the Processing icon in the taskbar. Odd.

DragonflyRobotics, 2022-03-13:
Interesting, could you provide the screenshots and the installers you used. I think I can help since I have had a similar issue before.

per1234, 2022-03-13:
>it displays the Processing icon in the taskbar

This issue is tracked at https://github.com/arduino/Arduino/issues/11744

per1234, 2022-09-24:
Thanks for your report @avlapp 

>It seems to be fixed in the new IDE install

That is great news. This means we can consider this resolved by the Arduino IDE 2.0.0 stable release.

closed 2022-09-24 by per1234