arduino/Arduino #11713: cant install old versions on linux

cybergrunge-ops, 2021-12-03:
why are versions prior to 1.0.6 not listed on teh website. this is FAKE if u arent listing all old versions. also imtrying to download 1.0.4 but that version has been CHANGED to where now there is no install or make file?????? how am i supposed to use it. please. i want version 1.0.4 or 1.0.1

kengdahl, 2021-12-16:
I have asked for the old versions to be added again. Waiting for a reply. 

Here they are in the meantime.
https://web.archive.org/web/20190426143829/https://www.arduino.cc/en/Main/OldSoftwareReleases

cybergrunge-ops, 2021-12-16:
both archive and current website, claims to provide compiled binary for Linux 32 and 64, but it doesnt contain a binary, just the source code. and as the issue says, current documentation - as well as what is in the archive link - does not provide instructions to build - there is simply a howto.txt which links to a page which ***no longer exist*** that supposedly has instructions on how to build arduino "on your favorite platform".

facchinm, 2021-12-16:
@cybergrunge-ops I don't know what you are talking about since the link that @kengdahl provided contains all the binaries as they were released at their time; scroll down the page, find `Arduino 1.0.x`, click  `Linux 32 bit`.
Links are in the format
http://downloads.arduino.cc/arduino-1.0.4-linux32.tgz or http://downloads.arduino.cc/arduino-1.0.4-linux64.tgz

Leaving the issue open until the links are restored here https://www.arduino.cc/en/Main/OldSoftwareReleases#1.0.x (for historical reasons only)

cybergrunge-ops, 2021-12-16:
Okay yes, i see via archive that they are how they were at time of release. 

It is just confusing because in 1.0.6(?) and onwards linux Arduino is presented in "portable" form, with an executable called arduino in the main folder along with hardware libraries folders etc, but when i download arduino-1.0.4-linux32.tgz and earlier, and open in, there is no executables inside, and the instructions inside point to a google code document that does describe building on linux, but only after v1.6.x onwards. User gets the feeling of being in some bizarre time warp you have no control over when you go to a link contained in a txt file for v1.0.4, and are suddenly transported to install instructions for v1.6.x .........  likewise the Linux Guide on the Arduino website only applies to v1.6.x onwards.

There are tons of projects and libraries out there that only work for 1.0.x, have no maintainers, but are still very useful, so i feel like this is a pretty important thing for arduino users to be able to do. I have installed 1.0.6 and it is mostly compatible with 1.0.4 projects. but i do consider this a bug in documentation as kengdahl tagged it, and one that is pretty crippling to users. gotta love "agile" software development...

per1234, 2022-05-14:
The fill list of Arduino IDE versions are now available from the "Previous IDE Releases" page (fixed back in February but we forgot to close the issue at that time):

https://www.arduino.cc/en/software/OldSoftwareReleases

Thanks so much for bringing this problem to our attention @cybergrunge-ops!

closed 2022-05-14 by per1234