arduino/Arduino #11715: Updating arduino car package in card manager purged the 3rd-party cards

thelwyn, 2021-12-06:
Today I was proposed to update the "Arduino AVR Boards" to version 1.8.4. After doing so, all the 3rd party cards installed have disappeared (I was using Huzzah Adafruit esp8266 breakut for example). 

When I go to Preferences , the field "URL for management of extra cards" was empty.

After writing down again the correct URL in this field, the card, reappeared, as "installed".

So actually what seemed to happen is that after updating, the "extra cards" field in Preferences had been erased, and because of this the extra cards were not visible anymore, even though they were still installed?

per1234, 2021-12-06:
Thanks for taking the time to submit a report @thelwyn.

>Arduino 1.8.8-Win 11

That version is 1.5 years out of date. We do not investigate defects in old versions of the Arduino IDE. Please try it again with the latest version of the Arduino IDE and then let us know whether or not the issue still occurs.

thelwyn, 2021-12-06:
Oh my bad, I'm closing the issue then. Will reopen if I experience it someday after updating to a recent version. Thanks for answering!

closed 2021-12-06 by thelwyn