arduino/Arduino #11728: Auto Format in menu Tools doesn't take effect

id1402, 2021-12-18:
Hello.
I have OpenSUSE Tumbleweed and this packages installed:

arduino-cli-fish-completion-0.18.3-2.51.noarch
arduino-core-1.8.3-2.88.x86_64
arduino-cli-0.18.3-2.51.x86_64
arduino-language-server-0.0+git.64a0e42-1.53.x86_64
arduino-cli-zsh-completion-0.18.3-2.51.noarch
arduino-builder-1.6.1-1.49.x86_64
arduino-mk-1.6.0-1.20.noarch
arduino-cli-docs-0.18.3-2.51.noarch
arduino-cli-bash-completion-0.18.3-2.51.noarch

Auto Format doesn't work.
Only error message output:

Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: cc.arduino.packages.formatter.AStyleInterface.AStyleMain(Ljava/lang/String;Ljava/lang/String;)Ljava/lang/String;
	at cc.arduino.packages.formatter.AStyleInterface.AStyleMain(Native Method)
	at cc.arduino.packages.formatter.AStyle.run(AStyle.java:80)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

per1234, 2021-12-18:
Hi @id1402. Thanks for your report.

Are you using the latest official version of Arduino IDE downloaded from this page?:

https://www.arduino.cc/en/software

If not, please download it from there and give that version a try. We don't provide support for issues that occur only in the 3rd party maintained unofficial package manager versions of the Arduino IDE.

per1234, 2021-12-18:
Ah, I see now that we already have a report about this, and it is indeed specific to the package manager version: https://github.com/arduino/Arduino/issues/5538

See that issue for all the details. If you have any additional information to add, you can comment there.

closed 2021-12-18 by per1234

id1402, 2021-12-20:
Hi per1234. Thanks for answer.

The packages were from here:
http://download.opensuse.org/repositories/CrossToolchain:/avr/openSUSE_Tumbleweed/x86_64/

I installed the software from https://www.arduino.cc/en/software now. Auto format works!
Everything seems to work. I will uninstall the Tumbleweed packages.

Only some WARNINGs in console output:
```
** (java:12965): WARNING **: 20:25:45.964: Invalid borders specified for theme pixmap:
        /usr/share/themes/Breeze/gtk-2.0/../assets/line-h.png,
borders don't fit within the image
** (java:12965): WARNING **: 20:25:45.964: invalid source position for vertical gradient
```

If the WARNINGs are unknown I can try to reproduce and describe.