arduino/Arduino #11736: ESP32 boards: `#include <avr/pgmspace.h>` results in `fatal error: avr/pgmspace.h: No such file or directory  -> candidates: []`

dmalenic, 2022-01-08:
```C
#include <avr/pgmspace.h>
```
compiles on all supported processors and architectures except ESP32. It is probably related to location of this file within the esp32 toolchain. The toolchain for all other processors and architectures (avr, arm, Intel and even esp8266) provide this file under path `.../packages/.../avr/pgmspace.h`, but esp32 toolchain provides it under path `.../packages/esp32/hardware/esp32/1.0.6/cores/esp32/pgmspace.h` which prevents compiling otherwise compatible sketches and libraries.

per1234, 2022-01-09:
> It is probably related to location of this file within the esp32

That is correct, and for this reason the error message is correct and expected. The esp32 boards platform is a 3rd party project not maintained by Arduino, so any discussion about the location is off topic here. I see it has already been discussed over at their repository: https://github.com/espressif/arduino-esp32/issues/508

closed 2022-01-09 by per1234

ktdp95, 2023-12-19:
Como posso resolver esse problema