arduino/Arduino #11739: >  **contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.**

19-6105zilleali, 2022-01-12:
>  **contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.**

This error was show in Arduino

_Originally posted by @19-6105zilleali in https://github.com/arduino/Arduino/issues/5377#issuecomment-1010600857_

per1234, 2022-01-12:
Hi @19-6105zilleali. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

I'll check in over there and if we do determine that the issue was caused by a bug in the Arduino IDE rather than the sketch actually having an unsupported encoding as the IDE claims, I'll be sure to reopen this.

closed 2022-01-12 by per1234

Rob58329, 2024-03-15:
I note that in Arduino IDE 1.8.19 weird characters in comments (eg. in "// Comment.." lines) also cause this "contains unrecognised characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload" warning message.

However doing a "Tools -> Fix Encoding & Reload"  does NOT fix this.
Instead you have to do the adruino.cc suggested: 
```
    Paste into Notepad
    Save as...
    Select "Encoding: ANSI" at the bottom of the "Save As" window
    Unrecognised characters will be converted to ?

```
(PS. I've posted this info here as I am unable to add any "reply"'s to any of the posts on arduino.cc.)