arduino/Arduino #11741: urgent wish for new features:  IDE 1.8.19: still no suitable search and replace functions, appearances, and features available

dsyleixa, 2022-01-12:
currently having used 1.8.9 but always being annoyed by the poor and inefficient search and replace functions and features  I now tried to upgrade to 1.8.19 - but it's still he same mess!
- the search popup windows is always positioned over the middle of the editor, always obscuring the editor code content!
- even when once moved to another place (e.g., at right side of the editor text window), this window does not store this new position but always appears again at the former position  over the middle of the editor when once pressing ctrl+f again for a new search:  totally annoying!
- there is still no way to search for "whole words only" but just features an entirely useless "Umbruch" (? linefeed"??) option.
- there still are not 2 different funcions available for "find and replace" (in other editors: ctrl+h) versus "(only) find" by ctrl+f without the risk of accidentally or incorrectly replacing keywords by former inputs. Even deadsimple editors like the MS-Windows Editor or Wordpad or IDEs like Geany got that feature!
- when once having entered a new search keyword which on wants to replace by another, still the old replacement is shown up and not even completely markes up in a whole, but the cursor then is placed behind the old replacement and one has to hit "Backspace" a million times to erase it before one can put in a valid new replacement expression; instead, when shifting to the replacement field, the entry had to be marked up in a whole, possibly just pressing "DEL" to erase it entirely at once.

when can that finally be fixed?

per1234, 2022-01-12:
>when can that finally be fixed?

It already is. You can download the Arduino IDE 2.0.0 release candidate now if you would like to give them a try:

https://www.arduino.cc/en/software#future-version-of-the-arduino-ide

closed 2022-01-12 by per1234

dsyleixa, 2022-01-12:
I actually don't want to test a probably bugged test release but download a final version.