arduino/Arduino #11742: Arduino IDE new install: no premature abort possible shortly after starting, either download or zip for an easily overlooked portable  option

dsyleixa, 2022-01-12:
when starting Arduino IDE new install: 
Arduino IDE new install: no premature abort possible shortly after starting, either download or zip for an easily overlooked portable  option.
OTOH, "downoad" is ambiguous because actually also the portable version is a "download".
So instaed there must be 2 differently termed install button links:
[X] download and install to the Windows Programs(0x86) folder
[X] download the portable version (zip file) to your harddisk

Additionally, there is a defciency in the installation program:
I actually wanted to install a portable option, but after mistakenly clicking the link "download", the installation to the Windows Programs(0x86) folder is started immediately, and after having recognized that it cannot be stopped anymore - just the extra features can be skipped.

So there must be  confirmation buttons before executing the installation:
[X] start installation to the Windows Programs folder
[X] Abort

per1234, 2022-01-12:
Hi @dsyleixa. Thanks for your suggestion.

>after mistakenly clicking the link "download", the installation to the Windows Programs(0x86) folder is started immediately, and after having recognized that it cannot be stopped anymore - just the extra features can be skipped.

It is not clear to me which link you are referring to.

Is it this one?

https://www.arduino.cc/en/software#download-options

[image]

dsyleixa, 2022-01-12:
yes, right side top, 
download options
Windows 7 and newer => [X] download and install to the Windows Programs(0x86) folder
Windows zip File => [X] download the portable version (zip file) to your harddisk

dsyleixa, 2022-01-12:
yes, right side top, but not the get app thing.

download options
Windows 7 and newer => [X] install to the Windows Programs(0x86) folder
Windows zip File => [X] download the portable version (zip file) to your harddisk

(no idea what a windows app [get] is all about though)

per1234, 2022-01-12:
OK, then that must be something you have configured in your operating system. You will need to change that configuration if you don't like it.

Those links only download the files. The "**Windows Win 7 and newer**" link does not do this:

> the installation to the Windows Programs(0x86) folder is started immediately

That only happens when you run the `.exe` file after downloading it.

---

>no idea what a windows app [get] is all about though

It is the Microsoft Store Windows App version of the Arduino IDE. It takes you to this page:
https://www.microsoft.com/en-us/p/arduino-ide/9nblggh4rsd8
and that page will indeed install the app, as is done with any app on the Microsoft Store.

closed 2022-01-12 by per1234

dsyleixa, 2022-01-13:
I never use  Microsoft Store Windows Apps why would one do that?
OTOH, of course the installation runs after clicking on the exe file, but then it starts immediately and cannot be stopped any more,
and before starting one is not even asked if and where / to which folder it shoulf be installed which version (portable or not).

=>  NOT SOLVED !!

dsyleixa, 2022-01-13:
I never use  Microsoft Store Windows Apps, what is that, what is it for, and  why would one do that?
OTOH, of course the installation runs after clicking on the exe file, but then it starts immediately and cannot be stopped any more,
and before starting one is not even asked if and where / to which folder it shoulf be installed which version (portable or not).

=>  NOT SOLVED !!

dsyleixa, 2022-01-13:
I never use Microsoft Store Windows Apps, what is that, what is it for, and why would one do that?
OTOH, of course the installation runs after clicking on the exe file, but then it starts immediately and cannot be stopped any more,
and before starting one is not even asked if and where / to which folder it should be installed by which version (portable or not).

And the explanations on the website are still ambiguos and confusing!!

=> NOT SOLVED !!

per1234, 2022-01-13:
>the installation runs after clicking on the exe file, but then it starts immediately and cannot be stopped any more,
and before starting one is not even asked if and where / to which folder it should be installed by which version (portable or not).

I'm not able to reproduce it. For me there is a standard [NSIS](https://en.wikipedia.org/wiki/Nullsoft_Scriptable_Install_System) installer, which has several dialogs allowing you to manage the installation:
[image]

[image]

[image]