arduino/Arduino #11744: Arduino IDE uses Processing IDE's icon when both are installed

autive42, 2022-01-20:
Expected Behavior:
When only arduino IDE is installed, the software shows up in the menu with the correct icon and when opened shows the correct icon on the panel.

After installing processing, this stops working correctly. The correct icon still shows up in the menu and can be pinned to the panel, but upon opening, opens with processing's desktop icon. The program otherwise works correctly. However, right clicking on the icon and selecting new window will open processing.

[Images](https://imgur.com/a/CZM23Aq)

OS and software:
Linux Mint 20.1
Arduino IDE 1.8.19
Processing 4.0

DragonflyRobotics, 2022-03-13:
I ran the installers in the same order you did. It worked for me just fine. They stayed separate and tilled too. Although, I am on Ubuntu 21.10, not Linux Mint. I will try that out too.
[image]

avlap, 2022-03-13:
I'm having this issue on Linux Mint 20 as well.

DragonflyRobotics, 2022-03-13:
Did you get any success with the .desktop file working? I am trying to get the Arduino icons to load. I already made the Arduino icons point to an absolute location and recompiled Arduino with no success.

DragonflyRobotics, 2022-03-13:
Also @avlap, you are using Linux mint 20.4 "Una" with Cinnamon UI, right? I am trying to exactly recreate your issue. Also, you ran the installers for Arduino and Processing twice right? I had to run it once with `sudo` and once without. This is because it is copying the files locally to your `.local` and `/usr/share` directories.

DragonflyRobotics, 2022-03-13:
Good news! It works for me in Linux Mint. 
[image]

avlap, 2022-03-13:
I was using Processing 3.

Running a installer twice, no I didn't do that and it doesn't sounds good tbh. That's a workaround at best isn't it?

I run a script, with these commands, where arduino/processing_destination is /opt
sudo tar -xvf "$download_destination/$tar_file" -C "$arduino_destination"
sudo "$arduino_destination/$app_name/./install.sh"

same for Processing:
sudo tar zxvf "$download_destination/$processing_file" -C "$processing_destination"
sudo "$processing_destination/./$app_name-$tag_name/install.sh" 

I do install Processing first and then Arduino.

DragonflyRobotics, 2022-03-13:
Gotcha, I am doing Processing 4 and Arduino first. Running it twice isn't exactly a work-around. I only do it if I want the desktop icon on my desktop. Otherwise `sudo ./install` works. I will try it your way and see.

DragonflyRobotics, 2022-03-14:
I was able to recreate the issue. I think that the make or path to the icon is getting mixed up. They have pretty similar installers.

per1234, 2022-09-24:
Thanks @autive42 This should be resolved in Arduino IDE 2.0.0. Please give it a try. The download links are listed here:

https://www.arduino.cc/en/software/

If the problem still occurs, you can submit a report with the updated details here:

https://github.com/arduino/arduino-ide/issues/new/choose

closed 2022-09-24 by per1234