arduino/Arduino #11753: [IDE V2] Introduce a separate metadata manifest file for each project

raffaeler, 2022-01-30:
I am currently using the IDE version 2.0.0 RC3 and this is a feature request.

## Proposal
It would be really useful introducing a separate **manifest** file that defines the **context** needed from `.ino` code to properly run.
- A list of boards that are compatible with the project
- For each defined board:
  - the board parameters for a specific variant, if any (voltages, bootloaders, programmer settings, etc.)
  - One or more compile time constants (to be used in preprocessor `#ifdef` macros) to map specific pins, include libraries, etc.
  - the last  known good serial port used to connect to the board
- All the metadata needed to properly define the `.zip` file defining a library (author, description, tags, etc.)
- External dependencies (libraries) specifying the exact version. There are often libraries depending on older versions of other libraries that do not work properly.
Probably a `.json` file would be the most appropriate choice.

Once the boards for a given project are clearly defined, the current combo box may just list those. And an additional combo box could instead specify the currently available physical ports where the board may be connected to.

## Issues addressed
- ability to work on different boards from different IDE windows
- ability to roam the code through the cloud on different PCs and still mapping the correct board
- matching the correct version of the external libraries
- clearly understand the dependencies of a project
- avoiding confusion in the community discussions: there are plenty of Arduino code examples on the internet that do not specify the board where the code can be run. This is often source of confusion and useless discussions.

Thank you for the great work.

per1234, 2022-02-01:
> introducing a separate manifest file that defines the context needed from .ino code to properly run.

The format of this file (`sketch.yaml`) has already been defined:

- Proposal: https://github.com/arduino/tooling-rfcs/pull/5
- Published RFC: https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md

>One or more compile time constants

This was intentionally left out of the initial proposal, but the desire for such a thing is well known and being tracked here:  https://github.com/arduino/tooling-rfcs/issues/9

>the last known good serial port used to connect to the board

This is a distinctly different class of information from what is intended to be stored in the `sketch.yaml` file, so I'm not sure it has a place there.

There is an existing [`sketch.json` sketch metadata file](https://arduino.github.io/arduino-cli/latest/sketch-specification/#metadata) used by [Arduino CLI](https://arduino.github.io/arduino-cli/latest/) to associate a port with a sketch.

There is also much previous discussion on this subject at

> the current combo box may just list those

Now we come to the subject of how the UI of the Arduino development tools interact with the information from the metadata file. Other than the use of the existing `sketch.json` data by [`arduino-cli board attach`](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_board_attach/) and [Arduino Web Editor](https://create.arduino.cc/editor), this is still not a settled question.

There has been much discussion of it here:

- https://github.com/arduino/Arduino/issues/1481
- https://github.com/arduino/Arduino/pull/7349
- https://github.com/arduino/arduino-cli/issues/108

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, since it looks like we have existing issues to cover the subjects you have raised, I'll go ahead and close this in favor of the others.

If you end up with additional information to share, feel free to comment in the appropriate thread.

closed 2022-02-01 by per1234

raffaeler, 2022-02-01:
@per1234 Glad to read that you are already working on these features. I tried to search similar issues with no luck.
Is the yaml file already in a nightly build or when will it comes?

per1234, 2022-02-02:
>Glad to read that you are already working on these features.

I'm also very happy that things have moved forward on this important system.

> I tried to search similar issues with no luck.

Even with a lot of familiarity with these repositories, I still had trouble assembling that collection of references. It is a complex subject because it was originally proposed back when there was only the Arduino IDE, but since then the project has been made more modular, meaning that the work, and requests for that work are split across multiple repositories.

>Is the yaml file already in a nightly build

No. Only the format of the file has been finalized via [RFC 0003](https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0003-build-profiles.md) ("Sketch build profiles and reproducible builds"). The implementation has not yet started.

The more limited [`sketch.json` sketch metadata file](https://arduino.github.io/arduino-cli/latest/sketch-specification/#metadata) system I mentioned has been in the release versions of [Arduino CLI](https://arduino.github.io/arduino-cli/latest/) for years though.

raffaeler, 2022-02-02:
Thank you, can't wait to see a pre-release with all of that!