arduino/Arduino #11754: Serial monitor @ the bottom bannar connected to the arduino icon

ilioSS53, 2022-01-31:
Hello,
Already many years I,am using arduino no problem so far.
But on version 1.8.13 I noticed that the serial monitor is working complete different and not usable at all.
It is stcked to the arduino icon in the bottom banner.
So I checked for a update and I updated it to 1.8.19 hoping the error is gone nope.
Looking for a solution and or similar reactons which i didn,t find I post this issue.

Windows 10
Like to hear to correct the serial monitor to have it working as earlier.

Regards,
Theo

per1234, 2022-01-31:
Hi @ilioSS53. Thanks for your report.

>It is stcked to the arduino icon in the bottom banner.

I'm not sure I understood the problem correctly. Please post a screenshot that shows the issue so I make sure I understand.

per1234, 2022-01-31:
Thanks. That is clear now.

Do you have multiple monitors? There is a similar report here: https://github.com/arduino/Arduino/issues/11337
And some additional ones at https://github.com/arduino/Arduino/issues/11256 and https://github.com/arduino/Arduino/issues/5581

ilioSS53, 2022-01-31:
Hi per1234,
Yes I use two monitors and I disconnected one. 
Started again windows and the computer and arduino.
So no result so far.
But I will look into the issues you mentioned.
Thanks so far I will check these issues.
Regards,
Theo

ilioSS53, 2022-01-31:
Hi per1234,
I followed the PaulStofregen proposal to delete the serial line from the preference file.
Still the same.
the preference file report  

run.display=1
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=false
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=115200
serial.line_ending=1
serial.parity=N
serial.port=COM6
serial.port.file=COM6
serial.port.iserial=null
serial.show_timestamp=false
serial.stopbits=1
sketchbook.path=D:\Arduino\Sketch-book

This means display only one I presume.

Started the computer no positive result?
I,am looking for the error so I can use the serial monitor again.
Regards,
Theo

per1234, 2022-01-31:
OK, thanks for the update!

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of https://github.com/arduino/Arduino/issues/11337.

If you end up with additional information to share, feel free to comment in the other thread.

If you would like to request support finding a workaround for the issue, please ask on the Arduino Forum. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/

closed 2022-01-31 by per1234