arduino/Arduino #11762: Arduino IDE should print out full arduino-builder command line when compiling and uploading

petaflot, 2022-02-15:
there are quite a few use cases where one prefers to use the command line version instead of the GUI. however, figuring the correct parameters to arduino-builder can be tricky.

a slightly more verbose output would be very appreciated, such as

`calling "arduino-builder -hardware <HARDWARE> -fqbn <FQ -compile`

per1234, 2022-02-15:
Hi @petaflot. Thanks for your interest in this open source project. I'm happy to tell you that this is already present in the verbose compilation output:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Check the box next to **Show verbose output during > ☐ compile**.
1. Click the <kbd>OK</kbd> button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for compilation to finish.
1. Scroll the output panel at the bottom of the Arduino IDE all the way up to the top.

There, you will see the `arduino-builder` commands.

For example:
```text
E:\arduino\ide\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware E:\arduino\ide\arduino-1.8.19\hardware -hardware C:\Users\per\AppData\Local\Arduino15\packages -hardware C:\Users\per\Documents\Arduino\hardware -tools E:\arduino\ide\arduino-1.8.19\tools-builder -tools E:\arduino\ide\arduino-1.8.19\hardware\tools\avr -tools C:\Users\per\AppData\Local\Arduino15\packages -built-in-libraries E:\arduino\ide\arduino-1.8.19\libraries -libraries C:\Users\per\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10819 -build-path C:\Users\per\AppData\Local\Temp\arduino_build_48204 -warnings=none -build-cache C:\Users\per\AppData\Local\Temp\arduino_cache_447367 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\per\AppData\Local\Temp\untitled980938763.tmp\sketch_feb15a\sketch_feb15a.ino
E:\arduino\ide\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware E:\arduino\ide\arduino-1.8.19\hardware -hardware C:\Users\per\AppData\Local\Arduino15\packages -hardware C:\Users\per\Documents\Arduino\hardware -tools E:\arduino\ide\arduino-1.8.19\tools-builder -tools E:\arduino\ide\arduino-1.8.19\hardware\tools\avr -tools C:\Users\per\AppData\Local\Arduino15\packages -built-in-libraries E:\arduino\ide\arduino-1.8.19\libraries -libraries C:\Users\per\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10819 -build-path C:\Users\per\AppData\Local\Temp\arduino_build_48204 -warnings=none -build-cache C:\Users\per\AppData\Local\Temp\arduino_cache_447367 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\per\AppData\Local\Temp\untitled980938763.tmp\sketch_feb15a\sketch_feb15a.ino
```

FYI, Arduino Builder is now deprecated. It is replaced by [Arduino CLI](https://arduino.github.io/arduino-cli/latest/), which provides all the capabilities of Arduino Builder and much more.

In fact, Arduino Builder is now only a wrapper around the Arduino CLI packages, serving as an adapter to the old Arduino Builder command line interface that is still used by Arduino IDE 1.x.

[Arduino IDE 2.x](https://github.com/arduino/arduino-ide) does not use Arduino Builder at all.

closed 2022-02-15 by per1234

petaflot, 2022-02-15:
that was fast :-)