arduino/Arduino #11768: Arduino IDE is not opening

arslan437, 2022-03-05:
IDE version : 1.8.19 

When I run the arduino debug exe file this is the what happen. 

[image]

DragonflyRobotics, 2022-03-11:
Could you please provide details regarding your installation and computer? I am assuming that you installed it using the installer and provided admin privileges. The only immediate problem I can think of is that the IDE can't initialize the Serial interface because it is lacking privileges. This happens to me in Linux when I suddenly unplug an Arduino and the whole program completely freezes.

arslan437, 2022-03-16:
I don't know what caused this. But I was working on esp32 based project. But suddenly it stopped working. I uninstalled the IDE and installed the latest version that I have downloaded from arduino.cc but didn't changed any thing.

I deleted the Arduino15 folder and then installed the IDE it worked after that.

closed 2022-03-16 by arslan437