arduino/Arduino #11770: function-definition is not allowed

geoahmad, 2022-03-14:
hello am new i Arduino and  I have a problem in the code of to control two 7segments using 2 shift registers
it telling me ,

a function-definition is not allowed here before '{' token
I don't know why.

const byte DATA_PIN=0;            //  Serial output to shift register 14
const byte LATCH_PIN=7;           //  shift register Latch pin 12
const byte CLOCK_PIN=8;           //   shift register clock pin 11

byte numbers[]={
  B11111100,     // 0
  B01100000,     // 1
  B11011010,     // 2
  B11110010,     // 3
  B01100110,     // 4
  B10110110,     // 5
  B10111110,     // 6
  B11100000,     // 7
  B11111110,     // 8
  B11110110,     // 9
  };

  
void setup() {
 // Set pins as output

  pinMode (DATA_PIN,OUTPUT);
  pinMode(LATCH_PIN,OUTPUT);
  pinMode(CLOCK_PIN,OUTPUT);

  void display_number(byte n){
  byte left,right;
  if (n < 10){
    digitalWrite(LATCH_PIN,LOW);
    shiftOut(DATA_PIN,CLOCK_PIN,LSBFIRST,numbers[n]);
    shiftOut(DATA_PIN,CLOCK_PIN,LSBFIRST,numbers[0]);  // make another number 0 to make shift
    digitalWrite(LATCH_PIN,HIGH);
  }
  else if( n>= 10){
    right= n % 10;
    left = n / 10;

    digitalWrite(LATCH_PIN,LOW);
    shiftOut(DATA_PIN,CLOCK_PIN,LSBFIRST,numbers[right]);
    shiftOut(DATA_PIN,CLOCK_PIN,LSBFIRST,numbers[left]);
    digitalWrite(LATCH_PIN,HIGH);
  }
}
  

void loop() {
  for(int i=0;i<100;i++)
    {
      display_number(i);
      delay(500);
    }

}

per1234, 2022-03-14:
Hi @geoahmad. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/programming-questions/20

closed 2022-03-14 by per1234

geoahmad, 2022-03-14:
sorry , for that.
ok I'll  go to Arduino Forum
thanks