arduino/Arduino #11774: IDE crashes when setting invalid Additional Boards Manager URLs

thorv, 2022-03-21:
### Describe the problem

The Arduino IDE crashes at the end of the splash.

### To reproduce

Set the Prefereces / Additional Boards Manager URLs to 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/
I know this is the wrong URL, but I still don't think the application should crash.

### Expected behavior

I'd like to be able to reconfigure the correct URL, in this case https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json, without the program crashing.
In addition, I think it would be better if the user was notified that the URL was incorrect.

### Arduino IDE version

Zip/Desktop 1.8.19

### Operating system

Windows

### Operating system version

11

### Additional context

I confirmed that the phenomenon is reproduced even in the Windows Sandbox environment.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-03-21:
Thanks for your report @thorv and for your assistance @PaulStoffregen!

I confirm this fault is only present in Arduino IDE 1.x and so I have transferred the issue to that project's issue tracker. @thorv there is no need for you to submit an additional report, but for future reference the issue tracker for the Arduino IDE 1.x-specific reports is here:

https://github.com/arduino/Arduino/issues

I can reproduce it on Windows, with a slightly different error message from the one @PaulStoffregen shared:

```text
java.io.FileNotFoundException: C:\Users\per\AppData\Local\Arduino15 (Access is denied)
        at java.io.FileInputStream.open0(Native Method)
        at java.io.FileInputStream.open(FileInputStream.java:195)
        at java.io.FileInputStream.<init>(FileInputStream.java:138)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:234)
        at cc.arduino.contributions.packages.ContributionsIndexer.mergeContributions(ContributionsIndexer.java:179)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:116)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
        at processing.app.Base.<init>(Base.java:272)
        at processing.app.Base.main(Base.java:150)
```

per1234, 2022-09-24:
As mentioned above, this bug only affects Arduino IDE 1.x, so is considered resolved by the release of Arduino IDE 2.0.0.

Note that there is a somewhat similar bug in Arduino IDE 2.x, which does not occur with the specific URL shared here, but does occur with invalid URLs that have a valid file name. That bug already has a dedicated issue with specific details so it is effectively tracked there: https://github.com/arduino/arduino-ide/issues/1084

closed 2022-09-24 by per1234