arduino/Arduino #11783: [Ubuntu 20.04] UTF-8 symbols not displayed properly in IDE

Batwam, 2022-04-14:
I am using code which includes UTF-8 symbols. These symbols are displayed fine on the Windows version but when I open the file in Ubuntu, most of them are not being displayed and there is not specific error generated in the terminal.

same issue here, some UTF-8 symbols are not showing like some of these: ☼🌞 
Worked fine on the Windows version but when I load up the file in Ubuntu, the characters are blank. As I long as I don't touch the file, the characters are still exported with the code but the IDE is not displaying them properly.

Is this a known issue with the IDE? I am using 1.8.15

#  File generated by update-locale
LANG="en_AU.UTF-8"
LANGUAGE="en_AU:en"

Windows version:
[image]

Ubuntu 20.04 version:

per1234, 2022-09-24:
Thanks for your report @Batwam. I am able to reproduce the bug with Arduino IDE 1.8.19, but not with Arduino IDE 2.0.0, so I will close this as resolved.

closed 2022-09-24 by per1234

Batwam, 2022-10-01:
confirmed, I just tried the 2.0.0 version in Flatpaj and this issue is fixed.