arduino/Arduino #11784: [Linux] Arduino config files should be saved in .config folder

Batwam, 2022-04-15:
Hello,

I am not sure if this has already been discussed and if there is a specific reason for this but the config files for the Arduino IDE in Linux should be saved in the $HOME/.config folder, not directly in the home folder.

# reference: 
https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html
[image]

per1234, 2022-04-15:
Thanks for taking the time to submit an issue @Batwam.

I see we have another report about this at https://github.com/arduino/Arduino/issues/3915.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-04-15 by per1234

Batwam, 2022-04-15:
@per1234 Thanks for pointing out the duplicate. I had a read through the other thread, it’s really a shame this issue isn’t being addressed even after people proposed patches.