arduino/Arduino #11788: Windows distribution: included Java Runtime 1.8.0_191 is a bit outdated

jpstotz, 2022-05-03:
Are there any plans to update the Java Runtime that is include din the Windows distributions of Arduino IDE?

At the moment (just tested the latest hourly build) Oracle Java 1.8.0_191 is included, a version that was released in October 2018. Since the release a few dozen vulnerabilities have been discovered in Java. I am not sure if Arduino is affected by one of those vulnerabilities but after more than three years an update would be really nice.

mrengineer7777, 2022-05-03:
Oracle's license changed on JDK 8 and up.  I suspect that's the issue.  A lot of dev tools I use are still on old versions of JDK 8.
https://www.oracle.com/java/technologies/downloads/#java8

jpstotz, 2022-05-03:
@mrengineer7777  I don't think that any of the special Oracle Java 8 are really needed (like Java WebStart), so switching to [Eclipse Temurin JRE/JDK](https://adoptium.net/temurin/releases/) should be an option.

per1234, 2022-09-24:
Closing as resolved since Arduino IDE 2.0.0 does not use Java.

closed 2022-09-24 by per1234