arduino/Arduino #11798: [Arduino IDE 1.8.19] Invalid Code Signature - expired code signing certificate

Bummibaer, 2022-06-16:
Is there are any news about certificates?
[[Arduino IDE 1.8.13] Invalid Code Signature - expired code signing certificate](https://github.com/arduino/Arduino/issues/10668)
AdafruitCircuitPlayground.cer :  12.01.2020 13:58
Arduino.cer : 18.01.2017 21:46
06.01.2019 01:59 : 06.01.2019 01:59

per1234, 2022-06-16:
Hi @Bummibaer. Thanks for your report and for referencing a previous issue.

The issue you referenced, https://github.com/arduino/Arduino/issues/10668, is about the certificate used to sign the Arduino IDE releases. You can see here that certificate was updated as promised and is not currently expired:

[image]

Yes, there is an upcoming expiration date, which we are aware of and will be sure to renew it.

I see that the certificates you are looking at are for the driver files bundled with the Arduino IDE. The situation with these is different, as has already been explained here: https://github.com/arduino/Arduino/issues/5762

>The drivers are timestamped, so the signature will still be valid even if the certificate used for signing expires, because the timestamp guarantee that the signature has been done before the certificate expiration.
The worst that can happen is that we can't sign new drivers, but this will produce an error early in the codesign tool during the build.

So the fact that they are expired is no problem at all for the currently distributed drivers. If there is a need to release new versions of the drivers, Arduino will be sure to take care to use certificates that are valid at the time those new releases are signed. So this is not something you need to be concerned about.

closed 2022-06-16 by per1234

cmaglie, 2022-06-17:
The signature of the IDE is timestamped as well, there is no need to re-release.