arduino/Arduino #11805: Arduino IDE deos not scale on a High DPI screen

ljden, 2022-07-16:
The Arduino IDE does not scale on a high DPI screen (resolution: 2560x1440)
Installed via `sudo apt install arduino` on Ubuntu 20.04 (apt thinks it installed version 2:1.0.5+dfsg2-4.1)
Arduino ide with terminal for scale:
[image]

per1234, 2022-07-16:
Hi @ljden. Thanks for your interest in this open source project.

> 2:1.0.5+dfsg2-4.1

That version is 9 years out of date. We do not support old versions of the Arduino IDE. A huge amount of work has been done on the IDE since then, including a complete rewrite in a different programming language for Arduino IDE 2.x. So it is likely that any bug experienced with that version is no longer relevant to the current development work.

Please always download the latest official nightly build of the Arduino IDE before submitting a bug report. We very much welcome all feedback for that version:

https://www.arduino.cc/en/software#nightly-builds

closed 2022-07-16 by per1234

ljden, 2022-07-16:
@per1234 hahaha all good. Do you guys manage the apt or is that a 3rd party? Any reason why it is so out of date?

matthijskooijman, 2022-07-16:
> Do you guys manage the apt or is that a 3rd party? Any reason why it is so out of date?

The official Debian packages are managed by Debian, not Arduino. They have gone without updates for years, due to a mix of technical, licensing and social issues, but now they are being updated again. However, your OS (Ubuntu 20.04) is two years old now, so it probably just missed the first update (Debian stable does have a more recent version).