arduino/Arduino #11810: Window 10 Monitor Copy fails (empty clipboard) when first char included in the selection

drmpf, 2022-07-22:
Tested using Arduino IDE 1.8.19 on Windows 10.
with this sketch
```
// Uno test for monitor copy/paste on Windows10
void setup() {
  Serial.begin(9600);
  Serial.println("Line 1");
  Serial.println("Line 2");
}
void loop() {
}
```
With AutoScroll OFF or ON
Ctrl+A  (shows both lines highlighted) followed by Ctrl+C and then Ctrl+V into notepad  => no text

With AutoScroll OFF or ON
Click and drag over both lines (showing both lines highlighted)
followed by Ctrl+C and then Ctrl+V into notepad  => no text

With AutoScroll OFF or ON
Click and drag over both lines **_BUT NOT the first char of the first line_** 
followed by Ctrl+C and then Ctrl+V into notepad  => 
```
ine 1
Line 2

```

EricTra, 2023-04-29:
It seems like you are experiencing an issue with copying text from the Serial Monitor in Arduino IDE to the Windows 10 clipboard. This issue occurs when the first character of the selected text is included in the copy. However, when the first character is not included, the text is copied successfully.

This issue may be related to **the way Windows handles the clipboard data when the first character is included in the selection**. One possible solution is to use a different text editor or serial monitor application that does not have this issue. You could also try copying the text to a different program first, such as Word or Notepad, and then copying it from there to your desired destination.

Another potential solution is to **update your Arduino IDE t**o the latest version or try reinstalling it to see if that resolves the issue. It is also possible that there may be a setting in Windows 10 that is causing this issue, so you could try looking into clipboard settings or resetting them to default.

-> Overall, this appears to be an issue that is specific to Windows 10 and the way it handles clipboard data. Hopefully, one of the above solutions will work for you and allow you to copy text from the Serial Monitor successfully.

per1234, 2023-05-09:
Thanks for your report @drmpf.

I just tested Arduino IDE 1.8.19 on my Windows 11 and Ubuntu 22.04 machines and was not able to reproduce the fault. So it seems it is specific to some conditions present on your machine but not mine (e.g., Windows 10 vs 11) rather than affecting all users universally.