arduino/Arduino #11811: Bug? VERY predictable output from random()

atom-smasher, 2022-07-26:
VERY predictable output from random(), after being seeded with an incrementing counter.

Attached sketch shows how I'm getting four streams of almost perfectly sequential numbers from an algorithm that uses multiple calls to random() after using an incrementing counter as a seed via randomSeed(). It should not be possible to get anything orderly from that, especially over such an immediately short period.

The sketch (below) produces mac-addresses as my original script does, then has four columns: The first three columns show sequential numbers from the high-byte of the last octet, the fourth column shows sequential numbers from the low-byte of the last octet.

A section of output:
```
     de:da:89:7e:e5:07    0         
     0a:7c:00:61:bf:b7            b
     36:1e:77:45:99:67        6     
     66:c0:ee:28:74:17    1         
     92:62:65:0c:4e:c7            c
     be:04:dc:ef:28:77        7     
     ee:a6:53:d3:03:27    2         
     1a:48:ca:b7:dd:d7            d
     46:ea:41:9a:b7:87        8        7
     76:8c:b8:7e:92:37    3         
     a2:2e:2f:61:6c:e7            e
     ce:d0:a6:45:46:97        9     
     fe:72:1c:28:21:47    4         
     2a:14:93:0c:fb:f7            f
     56:b6:0a:f0:d5:a7        a     
     86:58:81:d3:b0:58    5         
     b2:fa:f8:b7:8a:08            0
     de:9c:6f:9a:64:b8        b     
     0e:3e:e6:7e:3f:68    6         
     3a:e0:5d:61:19:18            1
     66:82:d4:45:f3:c8        c     
     96:24:4b:29:ce:78    7         
     c2:c6:c2:0c:a8:28            2
     ee:68:39:f0:82:d8        d        8
     1a:0a:b0:d3:5d:88    8         
     4a:ac:27:b7:37:38            3
     76:4e:9e:9a:11:e8        e     
     a2:f0:15:7e:ec:98    9         
     d2:92:8c:62:c6:48            4
     fe:34:03:45:a0:f8        f     
     2a:d6:7a:29:7b:a9    a         
     5a:78:f0:0c:55:59            5
     86:1a:67:f0:2f:09        0     
     b2:bc:de:d3:0a:b9    b         
     e2:5e:55:b7:e4:69            6
     0e:00:cc:9b:be:19        1     
     3a:a2:43:7e:99:c9    c         
     6a:44:ba:62:73:79            7
     96:e6:31:45:4d:29        2        9
     c2:88:a8:29:28:d9    d         
     f2:2a:1f:0c:02:89            8
     1e:cc:96:f0:dc:39        3     
     4a:6e:0d:d4:b7:e9    e         
     7a:10:84:b7:91:99            9
     a6:b2:fb:9b:6b:49        4     
     d2:54:72:7e:46:fa    f         
     02:f6:e9:62:20:aa            a
     2e:98:60:45:fa:5a        5     

```
That's seeded with 0x3ffe883c, and that output is near the beginning.

The first three columns to the right of the MAC addresses all show the high-byte of the last octet:
* The 1st column shows 0x0 - 0xf, sequentially
* The 2nd column shows 0x6 - 0x5, sequentially
* The 3rd column shows 0xb - 0xa, sequentially

The 4th column counts the low-byte of the last octet over a longer period, but the whole pattern repeats almost perfectly, and seemingly indefinitely, regardless of how it's seeded. Not the behaviour I expect from a PRNG!

nb the sketch has a delay() towards the bottom, so it scrolls up on a console at human-readable speed. For data-collection, just comment out the delay.

nb2 the random seed can be set manually.

sketch:
```
/*
  ===========================================

  buggy "random()" function, as observed here:
  https://github.com/atom-smasher/esp8266_beaconSpam

  this sketch starts with a random seed "randomMacSeed"

  that seeds the prng via "randomSeed()" and an incrementing counter
  
  calls to "random()" are producing some very predictable outputs!

  this sketch shows the mac-addresses as they would be generated,
  and derived from those mac-addresses are four columns, all counting
  incrementally, with very few errors

  expected results: using an incrementing counter to seed the prng, i expect
  to NOT have predictable output like this

  ===========================================
*/

const uint64_t randomMacSeed = os_random();     // random seed on startup
//const uint64_t randomMacSeed = 0x1234abcd ;   // fixed seed; make it your own

uint32_t i = 0;
uint8_t  macAddr[5];

void mayhemMac(uint32_t ssidNum) {
  randomSeed(uint32_t((randomMacSeed) + (ssidNum)));
  macAddr[0] = uint8_t(random(0x0, 0x100)) & 0xfe | 0x02 ;
  macAddr[1] = uint8_t(random(0x0, 0x100));
  macAddr[2] = uint8_t(random(0x0, 0x100));
  macAddr[3] = uint8_t(random(0x0, 0x100));
  macAddr[4] = uint8_t(random(0x0, 0x100));
  macAddr[5] = uint8_t(random(0x0, 0x100));
}

void setup() {

  // start serial
  Serial.begin(115200);
  //Initialize serial and wait for port to open:
  while (!Serial) {
    delay(10);
  }
  // wait for serial port to connect
  delay(300);
  Serial.println();

  Serial.printf("// Random Seed:    0x%x,    %d\n");

}

void loop() {

    for (i = 0; ; i++) {
      yield(); // needed for extra-large lists
      mayhemMac(i);
      Serial.printf("     %02x:%02x:%02x:%02x:%02x:%02x",
        macAddr[0], macAddr[1], macAddr[2], macAddr[3], macAddr[4], macAddr[5]);
        yield();

        if (0 == i % 3) {
          Serial.printf("            %x", macAddr[5] >> 0x4);
        }
        if (1 == i % 3) {
          Serial.printf("        %x     ", macAddr[5] >> 0x4);
        }
        if (2 == i % 3) {
          Serial.printf("    %x         ", macAddr[5] >> 0x4);
        }
        if (1 == i % 15) {
          Serial.printf("   %x", macAddr[5] & 0x0f );
        }

       Serial.println(); 
       delay(200);

       }
    }

```

nforystek, 2023-07-26:
I think they would have to incorporate a battery reach the global hardware scope for timers, or millis() through the break and revolution views where logically few or more never reach overflow between themselves thus the off state and on state are 2 of a pair to a virtual a 3rd party in global hardware scope continuance.

nforystek, 2023-08-08:
I just read Arduino has a 58-day limit in buffer overflow with the millis().  I always wonder how that goes. I imagine three internal tickers, one starting at zero, the other at the negative most number, and they both go forward equal same increment rage, and one starts at the positive most spectrum, and goes backwards at half the rate, all three then are part of calculation to issue a millis(), and because millis is never more than a second, you need then to get the calendar clocking segments like second and weeks, whereas when the timer reaching overflow threat approaches, it begins in reverse and the pri9or reversing one picks up in rate.  Something like that never has a second however, and then pending you can wait out how many years to possible catch the buffer overflow and it's a big Y2K mess!  But I say that because I think these timer states would be flash saved when booted down, or at a last state type of mode, then booting up the random seeding is partial to the system is unique in back powered up.  IDK, though.  Tuff one.  You could ignore real time and develop in the calendar intervals, second, minute, day, week, month and year.  Just say all Arduino start at a date when powered and kept from the user interaction.