arduino/Arduino #11815: Trojan:Win32/PackBackdoor ?

mm108, 2022-08-22:
I was trying to install [Arduino IDE 1.8.19](https://www.arduino.cc/en/software)  and as always I ran the installer through VirusTotal. One of the Security Vendor's Analysis detects Trojan:Win32/PackBackdoor.0142b83b. I am thinking that this could mostly be a false alarm? Any thoughts?

[image]

uttam2000, 2022-08-22:
Uttam

mm108, 2022-08-23:
> Hi @mm108. Thanks for bringing this detection to our attention. I have notified Arduino's security team of your report.

Thank you so much @per1234.

tofilwiktor, 2022-10-20:
When only one of the engines, especially a lesser-known one detects it, it is probably a false positive,

jguiherdev, 2023-02-03:
Hey guys, any new about this? was a false positive finally??

rhpco, 2023-02-03:
Hi @jguiherdev @per1234, 
After internal analysis, the security team concluded it was a false positive. 
We have contacted the vendor with a "false positive report" request, we hope the virustotal's engine will be updated quickly.