arduino/Arduino #11818: Normally, deleting the IDE should not erase your sketches.  They're stored in a separate location called the "sketchbook".

2dcdbrady, 2022-09-08:
Normally, deleting the IDE should not erase your sketches.  They're stored in a separate location called the "sketchbook".

However, perhaps you had the sketchbook location incorrectly configured to be the same directory as your Arduino installation?  This is already filed as issue #2719 

Could you look in File > Preferences and tell us what directory is configured for your sketchbook?  And of course, what directory is actually being used for Arduino?  I realize on Windows and modern Macs, many of the graphical programs for managing files try to "simplify" file systems by obscuring locations, so it may not be easy for you to get accurate info about where things really are on your computer's filesystem.  If File > Preferences, you can turn on verbose output while compiling, which will print compiler commands when you Verify or Upload.  Some of those will have the real, full pathnames.

If we're ever going to prevent this problem from wiping out anyone else's files in the future, we've really got to collect accurate info about exactly where you had things installed.  My hunch is this may be the already-known case of issue #2719, but really that's just a blind guess.

I've seen a couple other comments on the Arduino forum about something similar happening.  Maybe that was also you?  Or maybe this is a problem that's hitting multiple people?  I really want to figure out what happened and eventually add code to Arduino to make sure nobody else suffers this fate.  But until we understand why, the best I can say is maybe it's #2719, or maybe it's some other misconfiguration nobody has understood yet?  I know I've deleted and reinstalled Arduino many hundreds of times while developing features and testing, and I've never managed to stumble across this situation... so I really need details from you.

_Originally posted by @PaulStoffregen in https://github.com/arduino/Arduino/issues/2773#issuecomment-80757415_

per1234, 2022-09-08:
Hi @2dcdbrady. It is not clear to me what the purpose of this copy/paste issue is.

If you have additional information to share about the issue, please comment there so we can keep all related information consolidated in one place.

closed 2022-09-08 by per1234

2dcdbrady, 2022-09-08:
I know this is an old thread, but I couldn't find this anywhere else.
   I had not used my IDE for a while and when I returned and logged in and went to my sketches folder, they were all gone except for one, which only had my title and no code in it. I had only opened a couple of my sketches to run the last time I was there. and did not change or move anything around. I was fortunate to have screen shot some of what I was working on but most seems to be lost. 
  The format has been changed a bit so I am still looking and will check out the #2719 issue as well, but I do have one empty saved sketch. 
   Was there ever any more found on this issue?

2dcdbrady, 2022-09-08:
> I know this is an old thread, but I couldn't find this anywhere else. I had not used my IDE for a while and when I returned and logged in and went to my sketches folder, they were all gone except [...]

Thanks for putting up with me. I found them. For whatever reason, my icon in my tool bar no longer opened my IDE but only the web agent. I found the sketches in my documents and now opened the IDE app in finder and moved it to my dock.