arduino/Arduino #11819: Error message of Arduino IDE

Carpe-Wang, 2022-09-10:
When I used Arduino to burn esp32, I found that the error information was Java exception error information, such as null pointer exception. I want to ask if you can add some secondary error codes? It allows users to intuitively feel where the null pointer is reported in the IDE, which facilitates users to change the burning code in time.

gouriipawar, 2022-09-21:
Hey could  you please tell how can i reproduce this issue ?

per1234, 2022-09-21:
Hi all. The Arduino IDE project is no longer developed in the Java programming language so it will not be possible to accommodate this feature request.

Please give the new [TypeScript](https://www.typescriptlang.org/)-based Arduino IDE 2.x a try. Suggestions for enhancements to the new version are welcome.

https://github.com/arduino/arduino-ide

closed 2022-09-21 by per1234